ADVRHumanoids/XBotControl #5: Several crashes during Fresh Installation

dkanou, 2018-12-20:
1. Fresh Installation with Binaries and Superbuild with some config files
2. "set_xbot_config devel-superbuild/configs/ADVR_shared/user_example/centauro_simple_example.yaml"
3. "XBotCore -D" crashes
4. "XBOT_ROOT=$ROBOTOLOGY_ROOT XBotCore -D" solves the problem
5. gazebo crashes
6.  "GAZEBO_MODEL_PATH="" gazebo" solves the problem
7. "XBOT_ROOT=$ROBOTOLOGY_ROOT GAZEBO_MODEL_PATH="" roslaunch centauro_gazebo centauro_world.launch" crashes because it requires the HeriHand.so
8. We need to install the centauro_tools package, which is originally in master
9. This does not installa HeriHand
10. We switch centauro_tools to devel and then it is successful: "XBOT_ROOT=$ROBOTOLOGY_ROOT GAZEBO_MODEL_PATH="" XBotCore -D"
11. "rosrun cartesian_interface ros_server_node _use_xbot_config:=true" crashes and we cannot back-trace due to binaries.
11.

liesrock, 2018-12-20:
I believe there is a conflict of superbuilds: the exported global variables are the same so we should be sure you are actually sourcing ONLY the setup.bash from the binary.

Can you double check or try the installation on a brand new pc without any superbuilds?

dkanou, 2018-12-20:
When I use:
> set_xbot_config /opt/xbot/build/install/share/xbot/configs/ADVR_shared/coman/configs/coman.yaml
or
> set_xbot_config /home/dkanou/Kanoulas/code/devel-superbuild/configs/ADVR_shared/user_example/centauro_simple_example.yaml

this throws a seg fault:
> roslaunch centauro_gazebo centauro_world.launch

liesrock, 2018-12-20:
Let's go back at the setup: I believe the problem is that you have a superbuild + the binary. Is this the case? Can you try with a fresh system?

liesrock, 2018-12-20:
This superbuild:

/home/dkanou/Kanoulas/code/devel-superbuild

Should not be sourced otherwise it will go in conflict with the binary @alaurenzi

dkanou, 2018-12-20:
But then how can I use packages of the superbuild?  I mean centauro_simulation, etc?

liesrock, 2018-12-20:
I mention in an offline email that we should separate the two problem: let's first make sure the binaries work in a standalone fashion on your pc, we will then solve the issue you mentioned (sorry as I first user of both the superbuild and the binary the process is not super smooth).

Can you confirm the binaries are ok with COMAN from binary?

dkanou, 2018-12-20:
1. roscore
2. set_xbot_config /opt/xbot/build/install/share/xbot/configs/ADVR_shared/user_example/coman_example.yaml 
3. XBotCore -D
4.  roslaunch coman_gazebo coman_world.launch

Crashes:

dkanou@VisionPC:~$ XBotCore
[info] XBotCore using config file /opt/xbot/build/install/share/xbot/configs/ADVR_shared/user_example/coman_example.yaml
[warning] Error while trying to read the SCP log parameters: SCP log disabled
[warning] Yaml node IJoint missing in the config. file
[error] File libXBotEthernet.so NOT found
[error] Main catch exception: libXBotEthernet.so path must be listed inside LD_LIBRARY_PATH

dkanou, 2018-12-20:
Also: /opt/xbot/build/install/share/xbot/configs/ADVR_shared/coman/configs/coman.yaml is throwing a:

[info] XBotCore using config file /opt/xbot/build/install/share/xbot/configs/ADVR_shared/coman/configs/coman.yaml
[warning] Error while trying to read the low level config file: SCP log disabled
[error] Main catch exception: yaml-cpp: error at line 0, column 0: bad conversion
dkanou@VisionPC:~$ XBotCore -D
[info] XBotCore using config file /opt/xbot/build/install/share/xbot/configs/ADVR_shared/coman/configs/coman.yaml
[warning] Error while trying to read the low level config file: SCP log disabled
[success] DUMMY HAL interface found! 
[info] Spawned XBOTCORE thread
[info] Spawned COMMHANDLER thread
[error] in th_init! Config file does NOT contain mandatory node XBotRTPlugins!
[success] Thread XBOT_COMMHANDLER: start looping
ERROR in parseYAML : YAML file  /opt/xbot/build/install/share/xbot/configs/ADVR_shared/coman/configs/coman.yaml  does not contain XBotInterface mandatory node!!
Segmentation fault (core dumped)

dkanou, 2018-12-20:
Yes this runs.  But then when I start: roslaunch coman_gazebo coman_world.launch
it crashes everything (XbotCore)

liesrock, 2018-12-20:
So everything is working with the dummy (rviz) and the cartesian interface right?

I don't know about the roslaunch, but as usal for gazebo we just open the gzserver and gzclient and we load the robot (drag and drop it) with the plugin attached.

There is no need to have another XBotCore running and it is actually wrong.

So if you confirm the kinematic simulation is working we can then close the XBotCore instance (kill it) and proceed simply opening gazebo (no roslaunch to start) drag and drop the robot and try to control it

dkanou, 2018-12-20:
Ah wait.

> roscore
> XBotCore
> rosrun cartesian_interface ros_server_node

terminate called after throwing an instance of 'std::runtime_error'
  what():  robot_description_semantic parameter not set
Aborted (core dumped)

alaurenzi, 2018-12-20:
Ok..but then how to run a plugin that is compiled inside the superbuild? For example HeriHand inside centauro_tools

dkanou, 2018-12-20:
I see COMAN in superbuild just by running:

> roscore
> XBotCore -D

the cartesian interface crashes though...

liesrock, 2018-12-20:
Sorry @dkanou is the XBotCore run with -D? Why before it was working?

To answer to @alaurenzi this is something we did not designed: the idea is to release the binary of the plugins.
We can think about this, but first we should be sure that the binary are working in a standalone fashion.

dkanou, 2018-12-20:
Sorry Lu, I always run it with -D.

liesrock, 2018-12-20:
Ok so at least the RT plugins are ok right?
Can you run an HomingExample and see the robot moving on RViZ?

dkanou, 2018-12-20:
Yes homing for COMAN is working.

liesrock, 2018-12-20:
Ok super! Now regarding the cartesian interface are you using the 'use_xbot_config' option? Can you double check with @alaurenzi

dkanou, 2018-12-20:
> rosrun cartesian_interface ros_server_node _use_xbot_config:=true
Segmentation fault (core dumped)

liesrock, 2018-12-20:
Very strange, I am able to run it with no issue: are you able to provide more details about the crash?

I will try again on a fresh installation.

dkanou, 2018-12-20:
Eh no.  This is what we tried with @alaurenzi but we cannot back-trace the cartesian interface.

xinsongyan, 2018-12-21:
I installed XbotControl today, gazebo crash at start up without loading any robot.

Removing the "source /opt/xbot/build/install/share/xbot/setup.bash" line from bashrc, gazebo works properly. 

No idea why this is happening.

xinsongyan, 2018-12-21:
Solution: 

remove

`export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:$XBOT_ROOT/robots/iit-coman-ros-pkg/coman_gazebo/database`

from sourced xbot setup file: 

/opt/xbot/build/install/share/xbot/setup.bash

liesrock, 2019-04-23:
Hi @dkanou I am back: should be all solved in the latest version.

Available here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1

Documentation here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1

Can you give it a try?

liesrock, 2019-06-10:
@dkanou any chances to test this:

> Hi @dkanou I am back: should be all solved in the latest version.
> 
> Available here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1
> 
> Documentation here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1
> 
> Can you give it a try?

@nkashiri is also testing this. Otherwise I will close this and I kindly ask to open new issues if you find problems.

liesrock, 2019-06-19:
I assume it solved.

closed 2019-06-19 by liesrock