ADVRHumanoids/XBotControl #6: Library problem on fresh installation

edoardolamon, 2019-01-07:
After the installation of the binaries I set the config file of a robot and then ran XBotCore -D. I got the following error:

XBotCore: error while loading shared libraries: liborocos-kdl.so.1.3: cannot open shared object file: No such file or directory

liesrock, 2019-04-23:
Ciao @edoardolamon,

sorry for the late answer: I think this is related with your system packages, can you try the latest version of the framework?

Available here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1

Documentation here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1

liesrock, 2019-06-10:
@edoardolamon  any chances to test this:

> Ciao @edoardolamon,
>
> sorry for the late answer: I think this is related with your system packages, can you try the latest version of the framework?
>
> Available here: [...]

@nkashiri is also testing this. Otherwise I will close this and I kindly ask to open new issues if you find problems.

liesrock, 2019-06-19:
I assume it solved

closed 2019-06-19 by liesrock