ADVRHumanoids/XBotControl #7: The Centauro demo cannot be initialized correctly

XinyuanZhao, 2019-01-09:
I used the command `roslaunch centauro_gazebo centauro_world.launch` to launch the demo of Centauro. The urdf model could be loaded to Gazebo successfully. But when I started the simulation, the robot would sit and sprawl on the ground, which means that controllers from XBot plugins were actually not loaded correctly. Some config files in _ADVR_shared_ and _Centauro_Simulator_ should be modified to solve the above problem.

Assume we have set `set_xbot_config ADVR_shared/centauro/configs/config_centauro_full_body.yaml`. Possible solutions are listed below. 
1. In _ADVR_shared/centauro/configs/config_centauro_full_body.yaml_, add the command below in the front and delete everything after `GazeboXBotPlugin:`.
`XBotCore: config_path: ADVR_shared/centauro/lower/centauro.yaml`
2. In _ADVR_shared/centauro/srdf/centauro_full_body.srdf_, change the value of `tip_link` from _knee_i_ to _wheel_i_ (line 6, 10, 14, 18)
3. Create a folder _lower_ in _ADVR_shared/centauro/_ and create a config file _centauro.yaml_ (I suppose the one existing in superbuild works well here).
4. In _centauro/centauro_urdf/urdf/centauro.urdf_, change the type of joint _j_ft_1_ from _revolute_ to _fixed_.

Above are the major steps but not all that I took since there could be minor differences in working environments. For your information. By the way, if you think there is an easier way to fix this problem, please let me know. Thank you.

liesrock, 2019-04-23:
This should be solved in the latest version of the model released with XBotControl: can you try to install and test the v1.1.1 ?

Available here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1

Documentation here: https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1

liesrock, 2019-04-26:
Hi @XinyuanZhao, can you please test the new release and give us comments here? Your problem looks solved in other users experience.

liesrock, 2019-06-10:
Solved.

closed 2019-06-10 by liesrock