ADVRHumanoids/XBotControl #11: DEB file installation requirements

nkashiri, 2019-04-23:
Hi @liesrock 
I think it would be nice to have a list of requirements for the installation of the DEB file. 
I remember there used be to a basic script in the initial superbuild installation that would take care of prerequisites. Does the DEB file include such a thing? (I guess no, since I get a long list of dependency errors) 
Would you make a list of dependencies (or if possible, a similar basic script as we used to have)?

Thanks
Navvab

nkashiri, 2019-04-23:
I found the answer in https://github.com/ADVRHumanoids/XBotControl

nkashiri, 2019-04-23:
On the configuration section, I think it is better to explicitly mention adding the following line to .bashrc
. /opt/xbot/build/install/share/xbot/setup.bash
and then sourcing it by source ~/.bashrc

liesrock, 2019-04-23:
Thanks for the suggestions, should be now fixed in a better structure now

nkashiri, 2019-04-26:
At the "Quick Start" page, I believe we are missing the way of compiling and installing for the generated plugin

nkashiri, 2019-04-26:
at the "Basic Example" page, for running rviz, the following part:
_Remind to select an existent "Fixed Frame" from the list._
should be completed to:
_Remind to select "world" for "Fixed Frame" in "Global Options"._

nkashiri, 2019-04-26:
In the configure section of Installation page, I believe there is a repetition.

liesrock, 2019-06-19:
Thanks a lot @nkashiri, I close this: we can open new issues when you find problems in the documentation

closed 2019-06-19 by liesrock