ADVRHumanoids/XBotControl #12: set_xbot_config issue

nkashiri, 2019-04-23:
On setting the config file, I get:
$ set_xbot_config $XBOT_ROOT/configs/CentauroConfig/centauro_basic.yaml
/opt/xbot/build/install/bin/set_xbot_config: line 24: /home/navvab/.xbot/active_config.yaml: No such file or directory
/opt/xbot/build/install/share/xbot/configs/CentauroConfig/centauro_basic.yaml

The problem is the folder ".xbot" does not exist in my home directory.

@liesrock

liesrock, 2019-04-23:
Hi @nkashiri, thanks a lot! Are you sure you are sourcing bash file of the binaries?

nkashiri, 2019-04-23:
I think so, I added 
. /opt/xbot/build/install/share/xbot/setup.bash 
to .bashrc
and then sourced it by
 source ~/.bashrc
and I think it's fine, because I get the address by ecchoing the $XBOT.. 
Should I do something else?

liesrock, 2019-04-23:
You are doing everything good: the issue is related with the setup.bash, it is fixed in:

https://github.com/ADVRHumanoids/XBotControl/releases/tag/v1.1.1

liesrock, 2019-04-23:
Tested.

closed 2019-04-23 by liesrock