ADVRHumanoids/XBotControl #13: xbot gazebo controller

nkashiri, 2019-04-24:
Hi @liesrock 
The Gazebo doesn't get the Xbot Controller (with roslaunch centauro_gazebo centauro_world.launch paused:=false), the robot is not controlled. 
Do I need to change something?

liesrock, 2019-04-24:
I tested this using the centauro basuc config: which config are you using?

nkashiri, 2019-04-24:
I think I set the same file:
$ get_xbot_config 
/opt/xbot/build/install/share/xbot/configs/CentauroConfig/centauro_basic.yaml

nkashiri, 2019-04-24:
@liesrock: Looks related with the Gazebo version: 7 does not seems supported by the v1.1.1 of the binaries

liesrock, 2019-04-26:
It should be pointed out that the supported version is currently Gazebo 8: if you fix the issue upgrading it I will write this down explicitly in the documentation

nkashiri, 2019-04-26:
Yes, it works with the upgrade of Gazebo to version 8, and re-installation of ros and XBot. 
To avoid the removal and re-installation, I suggest adding the procedure to the installation section, here is what I did:
1 - Setup your computer to accept software from packages.osrfoundation.org.
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'

You can check to see if the file was written correctly. For example, in Ubuntu Xenial, you can type:
$ cat /etc/apt/sources.list.d/gazebo-stable.list
deb http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial main
2- Setup keys
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -

3- First update the debian database:
sudo apt-get update

4- follow instructions at https://medium.com/@abhiksingla10/setting-up-ros-kinetic-and-gazebo-8-or-9-70f2231af21a

closed 2019-04-26 by nkashiri