ADVRHumanoids/XBotControl #15: auto_generate_plugin compilitaion

nkashiri, 2019-05-09:
The auto generated plug-in does not include the catkin compilation configuration, may we have this?

liesrock, 2019-06-10:
I fixed the issue putting in the skeleton of the plugin the catkin_package declaration and adding the package.xml as requested by ROS.

I will release the new version of the xbotcontrol with these fix soon!

liesrock, 2019-06-11:
It can happen, based on user choice, that the catkin Package name does not follow the ROS naming conventions

nkashiri, 2019-06-12:
thanks @liesrock 
May you have a simple example in the wiki for this?

liesrock, 2019-06-12:
Hi @nkashiri this is the catkin tool usage

liesrock, 2019-06-12:
Anyhow I can include an example, even though you can use any build tool for the xbot plugins

nkashiri, 2019-06-12:
Yes, I can also take it from the ros catkin page, and copy paste it somewhere. But in my opinion, to have a user-friendly instruction, it's better to have the complete process in one place. A procedure said by the developer to follow one by one, and see an expected result. This what I would appreciate as a user.

liesrock, 2019-06-12:
So the point is that the user of the RT plugin generator is suppose to be a developer (i.e. skilled user), with basic knowledge of C++ and its related building/linking tools.

I fully agree with you about the user-friendliness, but a standard user will never write an RT plugin: for this category we should develop GUI and high level scripting tools.

Nevertheless, I will document one possible use-cases of the usage of XBotCore, documenting how to build a custom plugin with catkin

nkashiri, 2019-06-12:
I see your point, it's not meant for any user.
Yet, please consider that a skilled user for developing control schemes, while having beyond basic knowledge of C++, may not have the experience of dealing of ROS tools (although, I know it's becoming a trend now). Not every user is meant to be a computer scientist ;) 
I would say let's think of it as Simulink toolbox: when you install it, there is a test file that you can run it and realize if everything is fine, or no, you may get a message that you're missing some compiling library or some graphics driver...  
Anyway, even for a user with basic knowledge of C++, in my opinion, it gives a much better impression if the first run/test can come at no effort. This is just my perspective though :)

liesrock, 2019-06-12:
Exactly! For non computer scientist we need a tool like Simulink J

nkashiri, 2019-06-12:
So, I auto-generated plugin (generate_XBot_Plugin.sh Mytest) in my "~/catkin_ws/src"
then used "catkin_make" in the upper folder to compile it, and got the final message "[100%] Built target Mytest"
Next, source the setup.bash by ". /home/navvab/catkin_ws/devel/setup.bash"
But, when adding the "Mytest" to my config file, the launch fails

nkashiri, 2019-06-12:
@liesrock found an issue with sourcing when doing roslaunch
the temporary workaround is add to the bashrc the following:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/?????????????????/catkin_ws/devel/lib

liesrock, 2019-06-12:
Basically the issue is related to the sourcing of the catkin workspace which leads to an overwriting of the ROS_PACKAGE_PATH : looks like ROS is not giving much alternatives to this, except http://wiki.ros.org/catkin/Tutorials/workspace_overlaying @alaurenzi

liesrock, 2023-11-20:
I close this for inactivity

closed 2023-11-20 by liesrock