ADVRHumanoids/XBotControl #16: Eigen version in XBot

Yisoo-Lee, 2019-05-16:
I want to know the version of Eigen in XBot. Is it 3.2? I need to use some functions that are only available in Eigen 3.3 or higher, but I cannot use them. So I want to check it.

Yisoo-Lee, 2019-05-27:
I think XBot is based on Eigen 3.2. @liesrock, do you have any plans to upgrade to Eigen 3.3? Or is there an easy way to use Eigen 3.3 in XBotPlugin? I get an error when I run the plugin that includes Eigen 3.3.

alaurenzi, 2019-06-06:
Hi @Yisoo-Lee,
You could try to compile your own plugin with the latest Eigen3 version. However, this is not guaranteed to work since, as you noticed, XBotCore is compiled with 3.2.92 on our systems (i.e. the default version distributed with Ubuntu 16.04). If there is an ABI change between 3.2.92 and the latest version, you will obtain segfaults, memory corruption, and even wrong results without any error showing up.

Moreover, for us it is complicated to update the Eigen3 version, because some of our dependencies / third party software that we use may be compiled with 3.2.92.

By the way, which functionalities do you need that are not present in 3.2? Maybe we can find a workaround.

Yisoo-Lee, 2019-06-06:
Thank you @alaurenzi .
I want to use complete orthogonal decomposition ( https://eigen.tuxfamily.org/dox/classEigen_1_1CompleteOrthogonalDecomposition.html ).

alaurenzi, 2019-06-06:
I see.
As a workaround, you can create a separated library that wraps Eigen 3.3's COD class.
To be 100% sure about compatibility, the interface of this library should not be Eigen-based (e.g. you can use raw pointers to pass matrices around). However, it is pretty likely that basic Eigen types like MatrixXd WILL be compatible between 3.2 and 3.3, so they might be used in the interface as well.

Yisoo-Lee, 2019-06-06:
@alaurenzi, thanks a lot for your advice. I will then try to create a separated library as you suggested. I will let you know if I get any result.

closed 2019-10-08 by liesrock