ADVRHumanoids/XBotControl #17: Problems when sending torque references

Yisoo-Lee, 2019-05-21:
Hi @liesrock ,
It seems like sending torque reference through function "_robot->setEffortReference_" or "_model->setJointEffort_" doesn't work. (Both stiffness and damping are set to zero before using the functions.) 
I tested some cases with @EnricoMingo  to control joint angles with torque references, but all the results show weird behavior in the simulation.
When we check the log, unlike the reference torque in the plugin I created, the 'effort_reference' in XBotCore_log has different values.  As you can see in the following plots, 'effort_reference' in XBotCore_log has a constant value.

[image]
[image]

liesrock, 2019-05-21:
Hi Yisoo, thanks a lot for the detailed description: can you give me more information on how to reproduce the issue? 

I will try to run your plugin and check what you are saying.

Yisoo-Lee, 2019-05-21:
@liesrock , I sent the related material by email. Thank you!

Yisoo-Lee, 2019-05-22:
There was a mistake in my code. The problem is solved now.

closed 2019-05-22 by Yisoo-Lee