ADVRHumanoids/XBotControl #24: Contact Force Estimate

XinyuanZhao, 2019-07-26:
Hello @alaurenzi, I heard that you had written a function to estimate the contact force of Centauro. Could you please give me access to that function? Thank you very much.

alaurenzi, 2019-08-02:
Hi, have a look to this launch file!

https://github.com/ADVRHumanoids/centauro_cartesio/blob/master/launch/centauro_force_estimation.launch

XinyuanZhao, 2019-08-02:
Thank you very much.

I tried to launch this file but it popped up an error with the message " Invalid <node> tag: Invalid <param> tag: Unknown type 'yaml'. ". Then I deleted all the attributes " type="yaml" " from this file and launched it again, but there was still an error. I attached the whole message here.

Could you please help me to have a look at this issue? @alaurenzi 

> PARAMETERS
>  * /force_estimator/arm1_8/dofs: [0, 1, 2]
>  * /force_estimator/arm2_8/dofs: [0, 1, 2]
>  * /force_estimator/chains: [left_arm, right_...
>  * /force_estimator/is_model_floating_base: True
>  * /force_estimator/links: [arm1_8, arm2_8, ...
>  * /force_estimator/model_type: RBDL
>  * /force_estimator/wheel_1/dofs: [0, 1, 2]
>  * /force_estimator/wheel_2/dofs: [0, 1, 2]
>  * /force_estimator/wheel_3/dofs: [0, 1, 2]
>  * /force_estimator/wheel_4/dofs: [0, 1, 2]
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
> 
> NODES
>   /
>     centauro_cop_estimation (centauro_tools/centauro_cop_estimation)
>     force_estimator (cartesian_interface/force_estimation_node)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[centauro_cop_estimation-1]: started with pid [23214]
> process[force_estimator-2]: started with pid [23215]
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  robot_description_semantic parameter not set
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  robot_description_semantic parameter not set
> [centauro_cop_estimation-1] process has died [pid 23214, exit code -6, cmd /home/xinyuan/catkin_centauro_ws/devel/lib/centauro_tools/centauro_cop_estimation __name:=centauro_cop_estimation __log:=/home/xinyuan/.ros/log/c7c7542c-b52c-11e9-a420-54b20307e844/centauro_cop_estimation-1.log].
> log file: /home/xinyuan/.ros/log/c7c7542c-b52c-11e9-a420-54b20307e844/centauro_cop_estimation-1*.log
> [force_estimator-2] process has died [pid 23215, exit code -6, cmd /opt/ros/kinetic/lib/cartesian_interface/force_estimation_node ~torque_offset:=cartesian/torque_offset __name:=force_estimator __log:=/home/xinyuan/.ros/log/c7c7542c-b52c-11e9-a420-54b20307e844/force_estimator-2.log].
> log file: /home/xinyuan/.ros/log/c7c7542c-b52c-11e9-a420-54b20307e844/force_estimator-2*.log
> all processes on machine have died, roslaunch will exit

alaurenzi, 2019-08-02:
If you read the error message, it says `robot_description_semantic parameter not set`. This means that you need to upload the srdf to the param server under the given param name

XinyuanZhao, 2019-08-02:
Sorry to bother again. The error message at this time is
> [ INFO] [1564773546.060708644]: Private parameter ~/links is empty, exiting..
> ~LoggerClass
> terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
>   what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument

Do you have any ideas about how to fix this problem?

I have three more questions:
1. Do I have to complie the source file "centauro_addon.cpp" before I can successfully run the contact force estimation ?
2. The urdf and srdf I upload to ROS param `robot_description` and `robot_description_semantic` are obtained from the `CentauroConfig` package. Is it correct ?
3. The original launch file leads to an error " Invalid tag: Invalid tag: Unknown type 'yaml' ". Should I remove all the "type="ymal" from the launch file ?

liesrock, 2019-11-22:
I assume this was solved thanks to @alaurenzi suggestions: please reopen if I am wrong.

closed 2019-11-22 by liesrock