ADVRHumanoids/XBotControl #28: Dynamic memory allocation problem in Eigen3

rsubburaman, 2019-11-11:
I was running an RT plugin on CogIMon realtime pc, and it was repeatedly coming out of the real-time loop whenever it has to evaluate an expression that involved some matrix multiplications. This was happening even after predefining the size of all the matrices involved in that expression. The matrices size were no more than 15x15. After some browsing, I found that this is happening due to a temporary variable dynamically created by Eigen during such matrix multiplications. I also found the following link where a solution has been proposed to avoid this issue.

[https://github.com/stulp/tutorials/blob/master/test.md](url)

I tried the above solution, and it works. But the solution seems to make the code a little messy, especially if we have to evaluate a complex expression that involves several matrix multiplication operations. I was wondering if this issue has been addressed already and if the solution is better than the one proposed above.

liesrock, 2019-11-11:
Hi Raj,

yes as I already explained in person the problem is the temporary: @alaurenzi can tell you more about this.

alaurenzi, 2019-11-12:
Correct, matrix multiplications by default cause temporaries to be created. The reason is
 - possible aliasing (you can tell Eigen that there is no aliasing with the `noalias` method)
 - performance (`D = A*B*C` will always trigger a temporary for performance reasons)

Eigen3 is written to perform well in a general purpose scenario, and not to be real-time safe!

liesrock, 2019-11-14:
I close this, since it is related to a 3rd party library in a specific use case.

closed 2019-11-14 by liesrock