ADVRHumanoids/XBotControl #30: Comunication aborted for dropped messages

FrancescoRuscelli, 2019-12-19:
Sending commands to the robot after having changed the robot control mode back and forth (i.e. from Position to Stiffness to Position again) aborts the communication with the error: "Detected X dropped messages for joint ...". @alaurenzi

alaurenzi, 2019-12-19:
The bug is due to the xbotcore seeing a "hole" in the sequence identifier for the impedance fields of the control message. It is indeed a bug, because users are unable to change control mode during runtime. I'll think to a solution. Meanwhile you can comment the relevant lines on XBotNRTRef plugin so as to disable the check. @liesrock

liesrock, 2023-11-20:
Should be solved by @alaurenzi in xbot2.

closed 2023-11-20 by liesrock