ADVRHumanoids/XBotControl #32: Cartesio Function Undefined in Latest Version

christopherpeers, 2020-02-06:
Build fails due to the MakePostural() function being undefined in v1.1.8 of XBotControl. 
Can be avoided by removing cartesio but build still works when using an older version of XBotControl.

liesrock, 2020-02-07:
Can you give us an example of sources you are trying to build?

Thanks!

christopherpeers, 2020-02-07:
I believe I was building the master branch of the tutorial with XBotControl v1.1.8 installed. After I downgraded to the v1.1.4 installation, I had no issues building the tutorial.

Thanks

DavideAntonucci, 2020-02-07:
Hi Christopher,

Could you explain exactly which part of the tutorial you were performing and print out a screen-shot of the issue, please?

 Thanks,

Davide.

liesrock, 2023-11-20:
Following the new quickstart https://advrhumanoids.github.io/xbot2/master/quickstart.html the problem should not appear.

closed 2023-11-20 by liesrock