ADVRHumanoids/XBotControl #33: Removing Gazebo7 as written in wiki delete all ROS

torydebra, 2020-04-15:
Hi,

I followed the wiki for the installation. The required version of gazebo is the 8, but with kinetic the associated one is the 7. So, in the [wiki](https://github.com/ADVRHumanoids/XBotControl/wiki/Install,-Configure,-Uninstall) there is written to remove gazebo7. On my system, removing gazebo7 with apt-get will cause **all ROS package to be deleted**:
~~~
tori@Aspire-E15:~$ sudo apt-get remove gazebo7*
[...]
I seguenti pacchetti saranno RIMOSSI:
  gazebo7 gazebo7-common gazebo7-plugin-base libgazebo7-dev
  ros-kinetic-desktop-full ros-kinetic-gazebo-dev ros-kinetic-gazebo-plugins
  ros-kinetic-gazebo-ros ros-kinetic-gazebo-ros-control
  ros-kinetic-gazebo-ros-pkgs ros-kinetic-hector-gazebo-plugins
  ros-kinetic-simulators
~~~
(ros-kinetic-desktop-full is the "core" package of ros)
This seems a known problem ([here](https://answers.ros.org/question/313855/how-to-remove-gazebo-ros-pkgs/) for example) for any coupled version of ROS & Gazebo.

----
Related to this, from Gazebo wiki (and also [here](https://answers.ros.org/question/333395/can-i-properly-use-gazebo-8-with-ros-kinetic/)), it says:
> Warning!: Using this option, you won't be able to use any ROS Ubuntu package related to Gazebo from ROS deb repository. The equivalent of gazebo_ros_pkgs can be installed from debian packages, but all other software (such as turtlebot_gazebo) must be built from source.

So, it would not be better to indicate in the wiki how to install 2 different version of gazebo and how to make them coexist? (It seems possible from a quick look on internet)

liesrock, 2020-04-15:
Thanks @torydebra. Looks an important issue for the user: I know @DavideAntonucci worked on it in the past.

@DavideAntonucci can you have a look on this and let us know your opinion?

DavideAntonucci, 2020-04-15:
Hi guys (@torydebra , @liesrock ),

Unfortunately I removed gazebo 7 on my PC in order to have the right compatibility of our framework.
I didn't investigate about double consistency of versions.

So we need only to check and verify this opportunity. In the meanwhile @torydebra, if you have the possibility to install both packages but run gazebo 8, this could be great.

Later we can collect the right instructions and update the wiki.

Thanks,

Davide.

torydebra, 2020-04-15:
I will try to install both version. I will keep you updated

DavideAntonucci, 2020-04-15:
Thank you.

Davide.

torydebra, 2020-04-16:
The problem is harder than I think.

It seems impossible to have gazebo7 and gazebo8 together when you have also a ROS installation, when I try to install one the other is automatically removed.

_ros-kinetic-desktop-full_ (the recommended way to install ros) is a superpackage that install all, and the way most of the ros users installed it. 
With this one, among the others, gazebo7 is installed, so, if gazebo7 is removed singularly (as stated in this xbot wiki) the superpackage _ros-kinetic-desktop-full_ (with also some other packages related to gazebo7) is removed. This is not a big problem (at least I found no problem for now), because in truth ROS subpackages are still in the pc. 

The problem is that a lot of subpackages, without  _ros-kinetic-desktop-full_  are now in the "autoremove list":
~~~bash
I seguenti pacchetti sono stati installati automaticamente e non sono più richiesti:
  fltk1.3-doc fluid gazebo8-common libavdevice-dev libavfilter-dev libavresample-dev libfltk-cairo1.3
  libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev libfreeimage-dev libgts-dev
  libignition-msgs0 libignition-transport3 libpostproc-dev libprotoc-dev libprotoc9v5 libqtwebkit-dev
  libqwt-headers libqwt-qt5-6 libqwt-qt5-dev libsimbody-dev libtar-dev libzmq3-dev ros-kinetic-camera-calibration
  ros-kinetic-compressed-depth-image-transport ros-kinetic-compressed-image-transport ros-kinetic-depth-image-proc
  ros-kinetic-image-common ros-kinetic-image-pipeline ros-kinetic-image-proc ros-kinetic-image-publisher
  ros-kinetic-image-rotate ros-kinetic-image-transport-plugins ros-kinetic-laser-assembler
  ros-kinetic-laser-filters ros-kinetic-laser-pipeline ros-kinetic-perception ros-kinetic-perception-pcl
  ros-kinetic-polled-camera ros-kinetic-stage ros-kinetic-stage-ros ros-kinetic-stereo-image-proc
  ros-kinetic-theora-image-transport ros-kinetic-urdf-tutorial
~~~

I do not know which happen if I run autoremove. Probably then I have to reinstall them singularly, probably they are not really necessary, but I am scared to run it. 

**"Solution" which I found for now is:**
- remove gazebo7 and install gazebo8 (as stated in wiki)
- **DO NOT** run autoremove ever, until there is gazebo 8 instead of 7
- When a user want to switch back to gazebo7:
  ~~~bash
    sudo apt install gazebo7 libgazebo7 libsdformat4 libignition-math2
    sudo apt install ros-kinetic-desktop-full
  ~~~
  The first command remove gazebo8 and install gazebo7.
  The second is to remove all the packages from the "autoremove list"

Sorry if this post is confusionary, it is so also for me. 
Another method can be to use [dockerhub](https://hub.docker.com/_/gazebo), maybe with this is possible to have 2 gazebo versions.

torydebra, 2020-10-29:
Found that gazebo can be compiled in a catkin_workspace [link](http://gazebosim.org/tutorials?tut=install_from_source#Installinacatkinworkspace). This may permit to have multiple gazebo version, choosing the right one by sourcing the devel of the right catkin space. Maybe this is useful for someone (included future me)

liesrock, 2023-11-20:
On 20.04 everything should be good: I'll close.

closed 2023-11-20 by liesrock