ADVRHumanoids/XBotControl #37: Using XBotCore and CartesI/O with a real robot

vsimundic, 2021-04-26:
Hi,
I am currently working on ABB-IRB 2400L robot arm for which I have srdf files, urdf files and I made all necessary config files according to the tutorials and instructions. I successfully set up the simulation and now I'm trying to find a way to connect everything to the real robot. I'm using XBotCore version 1.1.8 and when I run `$ XBotCore` (not the dummy one) it fails with:
```
[error] File libXBotEthernet.so NOT found
[error] Main catch exception: libXBotEthernet.so path must be listed inside LD_LIBRARY_PATH
```
Do I have to write my onw libXBotEthernet or am I not seeing something here?
Any help would be appreciated.

liesrock, 2021-05-20:
Hi @vsimundic: sorry for the late reply. We are currently working on a newer version of the framework which will speed up the integration of a new robot also on the real hardware.

I will keep you updated!

vsimundic, 2021-05-21:
> Hi @vsimundic: sorry for the late reply. We are currently working on a newer version of the framework which will speed up the integration of a new robot also on the real hardware.
> 
> I will keep you updated!

That would be great. Thank you for answering!

liesrock, 2023-11-20:
Please try to refer to: https://advrhumanoids.github.io/xbot2/master/quickstart.html

closed 2023-11-20 by liesrock