ADVRHumanoids/XBotControl #46: [CartesIO] Reach a custom joint configuration in cartesio gui without simulation or real robot (visualization only)

IoannisDadiotis, 2021-07-16:
@alaurenzi I want to work on visualization (no simulation or real robot) of Centauro with the backward arm configuration. I have handcrafted a sequence of motions in the joint space to move from homing to a custom backward-arms configuration. I am successfully implementing this in the physics-based simulator through publishing in the `/xbotcore/command` topic.

However, I cannot understand how to load this custom joint configuration in Cartesio when simulation is not running. Through the `/cartesian/Postural/*` topics it seems that the user is able to send commands in the joint space, but these commands are affected by the type of hierarchical WBC that is used (thus when commanding one joint, other joints may also be moved as a result of the WBC). Ideally, I would like to be able to move only the joints that I command so that I can reach my custom configuration. Another solution would be to load somehow my custom joint configuration as the initial configuration of the robot when launching cartesio in visualization only. Is there a way to achieve this?

alaurenzi, 2021-07-19:
Hi @IoannisDadiotis, you should be able to change the starting posture for the robot via the `/cartesian/home` parameter. For instance, you can `rosparam set /cartesian/home/j_arm1_4 1.0` before launching cartesio to have all the robot in zero position except the left elbow at q = 1.0 rad. This is done more efficiently with a yaml file where you store the starting posture as a joint name -> joint pos map, and then `rosparam load myhome.yaml /cartesian/home` as [explained here](http://wiki.ros.org/rosparam#rosparam_load).

IoannisDadiotis, 2021-07-19:
Thank you. It works.

closed 2021-07-19 by IoannisDadiotis