ADVRHumanoids/XBotControl #49: Time logging in autogenerated ros_executor* files

IoannisDadiotis, 2021-09-27:
When running cartesian interface log files are automatically generated as `.mat` files in the `/tmp` folder. `ros_executor*` files provide structures for the joint trajectories (position and velocity) but there is no information about the absolute time that the joint properties are sampled, thus the user has to know beforehand the frequency at which the interface is running in order to acquire the joint trajectories wrt to time.

The variables `run_time` and `ros_time` that are saved are not related with this. These consist the time needed for solving the IK and updating the IK solution (or something similar), respectively, according to @liesrock. 

I suggest to augment this log file with a variable for the absolute time so that the user can reconstruct the trajectories with only the log file at hand and no prior knowledge of the logging frequency.

liesrock, 2023-11-20:
@IoannisDadiotis did you add this feature ?

IoannisDadiotis, 2023-11-20:
@liesrock, I didn't add this. I have binaries on my machines, so it is not so straightforward to do it.

closed 2023-11-20 by liesrock