ADVRHumanoids/XBotControl #51: avoiding absolute path

lia2790, 2021-11-08:
Hi all,

I am trying to avoid absolute path in xbot control plugin passed as a parameter as mentioned [here](https://github.com/ADVRHumanoids/VariableImpedanceRegulation/issues/23#issue-1046529835).

Basically, I want to change the code below

```yaml
joint_impedance_control: 
        thread: rt_main
        type: joint_impedance_controlH
        params:
            urdf_path: {value: /home/liana/catkin_ws/src/VariableImpedanceRegulation/robots/6dof/urdf/6dof.urdf, type: string}
            srdf_path: {value: /home/liana/catkin_ws/src/VariableImpedanceRegulation/robots/6dof/srdf/6dof.srdf, type: string}
            task_path: {value: /home/liana/catkin_ws/src/VariableImpedanceRegulation/robots/6dof/task/6dof_stack.yaml, type: string}
``` 

into the following

```yaml
joint_impedance_control: 
        thread: rt_main
        type: joint_impedance_controlH
        params:
            urdf_path: {value: $(rospack find vir_singlejoint)/robots/6dof/urdf/6dof.urdf, type: string}
            srdf_path: {value: $(rospack find vir_singlejoint)/robots/6dof/srdf/6dof.srdf, type: string}
            task_path: {value: $(rospack find vir_singlejoint)/robots/6dof/task/6dof_stack.yaml, type: string}
``` 

in such a way to have a more portable code.

I tried this but I obtain an error as below
```
liana@iitadvrlw003 ~/catkin_ws/src/VariableImpedanceRegulation/configs/xbot (cartesIO)$ ./setup_xbot2_gz_6dof.sh 
/home/liana/catkin_ws/src/VariableImpedanceRegulation/configs/xbot/robot/6dof.yaml
[warn][jm] running with disabled safety policies 
[ok  ][xbot2-core] started running 
Initializing CartesIO Interface...
uploading models urdf srdf...
[err ] in generate_jidmap: urdf does not exists! 
generating jidmap...
setting model type...
[ERROR] [1636368178.756491295]: Error document empty.
``` 
[image]

**To replicate the error**

1. clone the repo [here](https://github.com/ADVRHumanoids/VariableImpedanceRegulation)
2. go to the cartesIO_debug branch
3. go to the commit ISSUE_ABSOLUTE_PATH 30bdc4dc476adb6b4a32db7ac8d8fa9e0fc1dfa8
3. run `roslaunch vir_singlejoint vir_6dof_gazebo.launch` in a terminal
4. run  `./setup_xbot2_gz_6dof.sh` in other terminal
5. then you should see the error reported.
(source the terminals)

liesrock, 2021-11-22:
@lia2790 did you tried @alaurenzi suggestion? Can we close this?

lia2790, 2021-11-22:
Hi, yes I tried again with the lastest version of xbot2 (xbot2-core  version: 2.2.0 (734fd97)) and it is still not working

[image]

alaurenzi, 2021-11-22:
Did you try to mimic the snippet I provided ?

lia2790, 2021-11-22:
sorry yes now i tried and I have the same issue, it does not take the path correctly... I am creating a proper branch in the repo VariableImpedanceRegulation in order to solve this problem as luca suggested

lia2790, 2021-11-22:
## How to reproduce the issue

1. go to the repo VIR ([here](https://github.com/ADVRHumanoids/VariableImpedanceRegulation))
2. download it or clone it
3. go to the branch ** xbot_absolute_path_failing_issue#51**
4. then run the vir with the following commands, each one in different terminals and before to lunch them run "source devel/setup.sh"
4a. run roslaunch vir_singlejoint vir_centauro_gazebo.launch
4b. go inside the folder ~/catkin_ws/src/VariableImpedanceRegulation/configs/xbot and run  ./setup_xbot2_gz_centauro.sh 
4c. run  rosservice call /xbotcore/joint_impedance_control/switch 1

alaurenzi, 2021-11-22:
Does `type: file` work?

lia2790, 2021-11-22:
This I did not try, are both of them equal?

lia2790, 2021-11-22:
Anyway I tried but I obtained a strange error, I mean It returns some errors which seem that some params have not been read as supported to. I mean with the all path I did not get any error and the code works as expected

alaurenzi, 2021-11-22:
Can you report the error here?

lia2790, 2021-11-22:
Yes, I obtain the following errors continuously

``` 
ERROR in setDamping : D has wrong size 7 != chain joint number 0
ERROR operator() : you are requesting a chain with name arm that does not exists!!
ERROR in setStiffness : K has wrong size 7 != chain joint number 0
ERROR operator() : you are requesting a chain with name arm that does not exists!!
ERROR in setDamping : D has wrong size 7 != chain joint number 0
ERROR operator() : you are requesting a chain with name arm that does not exists!!
ERROR in setStiffness : K has wrong size 7 != chain joint number 0
ERROR operator() : you are requesting a chain with name arm that does not exists!!
ERROR in setDamping : D has wrong size 7 != chain joint number 0
ERROR operator() : you are requesting a chain with name arm that does not exists!!
ERROR in setStiffness : K has wrong size 7 != chain joint number 0
ERROR operator() : you are requesting a chain with name arm that does not exists!!
``` 

I push it in the last commit **80249e5180ab29ca1988c6a9153068103840e8a3**

lia2790, 2021-11-22:
My computer was blocked due to gazebo simulation then I switched it off, re-started... I did all the changes required (add the print to check the string and compile it) ... then the code works with **type:file** ... very strange but actually it works so we can close this issue

closed 2021-11-22 by lia2790